weltenwort/home-assistant-rct-power-integration #91: Support for the new "Energy HA integration"

pluskal, 2021-08-05:
**Is your feature request related to a problem? Please describe.**
Home Assistant announced a new way to monitor Energy. In order to integrate with that, appropriate sensors need to be configured for [Long-term Statistics](https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics).

**Describe the solution you'd like**
Set sensor `state_class` to `measurement` for sensors monitoring current `production, consumption, grid, battery, etc.`

**Describe alternatives you've considered**
None

**Additional context**
[Home Assistant 2021.8 Announcement for Home Energy Management](https://www.home-assistant.io/blog/2021/08/04/home-energy-management/)

weltenwort, 2021-08-05:
@pluskal thanks for filing this issue! The energy integration looks exciting and I'll investigate supporting it as soon as I find the time.

weltenwort, 2021-08-07:
I tried to implement this in #98 but it takes a while for the statistics to become available. I'll keep an eye on it over the next days.

If you are a user of this integration (or a developer of a similar one) I'd be interested in hearing whether it works for you.

weltenwort, 2021-08-07:
Looks good so far:

[image]

Apparently one has to wait at least a few hours for values to show up.

pluskal, 2021-08-08:
Nice! You made it very, very fast :). Thanks!

It seems to work flawlessly.

[image]

weltenwort, 2021-08-08:
Something might not be right with the grid production. It stays fixed a 0 kWh even though it's feeding to the grid. Unfortunately there is no documentation about whether the different numbers are expected to be positive or negative. I'll have to dig into the HA core code to find that out. :eyes:

pluskal, 2021-08-08:
I think that It is ok in my case. I have disabled `grid production` in the RCT application for now.

weltenwort, 2021-08-08:
There seem to be quite a lot of confusion about the way the energy dashboard works. I'll wait until it has settled a bit and until more built-in integrations support it so I can learn from them. Either that or until a proper documentation exists for how energy sensors interact with the energy dashboard.

I'll reopen the issue in the meantime in order to be transparent about the limitations.

k3mpaxl, 2021-08-22:
The reason why grid production is not working well is that its a negative value. I solved this issue with these few lines:

```
template:
  - sensor:
      - name: "RCT Grid Energy Production Day Inverted Value"
        unit_of_measurement: "Wh"
        state_class: "measurement"
        device_class: "energy"
        state: >
          {% set production = states('sensor.rct_power_storage_grid_energy_production_day') | float %}

          {{ (production * -1) |round(1) }}
        attributes:
          last_reset: "{{ state_attr('sensor.rct_power_storage_grid_energy_production_day', 'last_reset') }}"

```

[image]

weltenwort, 2021-08-23:
@k3mpaxl thanks for sharing that. It's what I almost suspected. Maybe I should add a positive alternative for the grid feed sensor.

hanzsez, 2021-09-08:
is there something extra which has to be added/manualy modified to make this intergration work with the Energy panel? I installed clean home assistant in docker, manually added the integration, added the invertor with bateries so they correctly show all the possible values. But even after several hours when I try to enroll to the energy meter data, it tells me there are no data source listed to pick the statistics from.

k3mpaxl, 2021-09-08:
Home assistant changed from the state requirements from measurement to total_increasing in version 2021.09.02.

Since a few days the integration got updated and there are total entities that can be selected:

[image]

weltenwort, 2021-09-08:
Yes, the fact that the inverter reports the produced energy with a negative sign is inconvenient. The refactorings I did as part of #120 will make it easier to add "calculated" entities. I'm planning to expose an entity for the absolute value of the produced energy with the correct sign as well as the sum of the energy produced by the two generators. Hope I'll get around to it this weekend.

closed 2021-09-17 by weltenwort

weltenwort, 2021-09-17:
I just released [v0.8.1](https://github.com/weltenwort/home-assistant-rct-power-integration/releases/tag/v0.8.1), which includes a new entity `"Grid Energy Production Absolute Total"` that has the sign the energy dashboard expects. I left the original entity in place for backwards-compatibility.

PR #98 (2021-08-07): Opt into new long-term statistics feature to support improved energy analytics
## :notebook: Summary

This enables the new long-term statistics features for the relevant sensor entities in order to enable their usage on the new energy dashboard. It also upgrades some dependency versions.

closes #91

## :detective: Implementation notes

This improvement has some quirks that should be improved upon later:

- It indiscriminately adds the `SensorEntity` into the entity inheritance chain to support the new sensor properties. Since this integration doesn't use any other platforms yet, this should b [... cut]
PR #127 (2021-09-17): Add calculated entities for power generation
## :memo: Summary

This adds three new entities that are calculated from the measurements:

- `"Grid Energy Production Absolute Total"`, which is the absolute value of the energy fed into the grid for usage with the energy dashboard
- `"All Generators Energy Production Total"`, which is the sum of the energy produced by generators A and B for simpler use in the energy dashboard
- `"All Generators Power"`, which is the sum of the instantaneous power of generators A and B

closes #91
closes #119

## :female_detective: Implementation notes

The followi [... cut]