weltenwort/home-assistant-rct-power-integration #100: Round Response Values

k3mpaxl, 2021-08-09:
Hi @weltenwort ,

thank you very much for your effort. I'm pretty impressed by the amount of time and # of codelines you've put into this integration. I already found rctclient and was happy that you already figured out how to integrate the rct power inverter into home assistant.

However, the response values seem to be a bit long. I'd like to suggest to have them rounded like most other sensors do (one digit after the decimal separator).

Is it possible to round the values? a simple "round(decoded_value,1)" didn't do the trick.

Thanks in advance!
Have a nice day.

weltenwort, 2021-08-09:
Thanks for bringing that up. I agree that the precision is too high for some values. I'll look into limiting it where possible.

Originally I was hoping that Home Assistant would provide a way to define the precision when rendering. But by now I think it might be worth limiting the precision earlier during the ingestion process, because it could reduce the size of the recorded history when the values fluctuate less.

weltenwort, 2021-08-12:
I think a fixed precision for rounding would be fine for now if chosen appropriately. A PR would certainly be welcome :heart: it might be a few days until I get around to reviewing it, though :innocent:

k3mpaxl, 2021-08-22:
As you've got multiple branches, I decided to not create the PR. However this is the working section within "lib/entity.py":
```
    @property
    def state(self):
        """Return the state of the sensor."""
        value = self.get_valid_api_response_value_by_id(self.object_ids[0], None)

        if isinstance(value, bytes):
            return value.hex()

        if isinstance(value, tuple):
            return None

        if isinstance(value, Number):
            if self.unit_of_measurement == "%":
                value = value * 100

            return round(value,1)

        return value
```

weltenwort, 2021-08-28:
@k3mpaxl thank you for the snippet. I implemented it in #111, so it will be included in the next release, which I'll hopefully cut today.

PR #111 (2021-08-28): Round numeric state values to one decimal place
## :notebook: Summary

This rounds numeric state values to one decimal place.

closes #100
closed 2021-08-28 by weltenwort