weltenwort/home-assistant-rct-power-integration #134: Allow network frequency monitoring without island mode capabilities.

Hi,
It would be wonderful, if you could integrate the frequency monitoring value that is displayed within the RCT App.

It is displayed under "Messwerte" - > "AC" - > "Netzfrequenz"

k3mpaxl, 2021-09-23:


weltenwort, 2021-09-23:
Hi @k3mpaxl, great suggestion - thank you. Am I correct to assume that "without island mode capabilities" means there is no power switch in the system?

It takes some guessing to map the values displayed in the App to register values. My first guess would be that this frequency is `grid_pll[0].f`. :thinking: I'll give that a try, but if anyone knows better please let me know.

k3mpaxl, 2021-10-01:
Hi @weltenwort ,

I can confirm that this is the correct datapoint. Just compared the values from the android app and the ones fetched with rctclient.
Have a nice day!
k3mpaxl

k3mpaxl, 2021-12-19:
Hi @weltenwort,

I hacked the frequency reading into your component and would like to show you how I did it:

In `lib/state_helpers.py` I modified the round behaviour to have a more accurate precision:
```
def get_api_response_value_as_state(
    entity: SensorEntity,
    value: Optional[ApiResponseValue],
) -> StateType:
    if isinstance(value, bytes):
        return value.hex()

    if isinstance(value, tuple):
        return None

    if isinstance(value, (int, float)) and entity.native_unit_of_measurement == "%":
        return round(value * 100, NUMERIC_STATE_DECIMAL_DIGITS)

    if isinstance(value, (int, float)) and entity.native_unit_of_measurement == "Hz":
        return round(value, 3 * NUMERIC_STATE_DECIMAL_DIGITS)

    if isinstance(value, (int, float)):
        return round(value, NUMERIC_STATE_DECIMAL_DIGITS)

    return value
```

In `lib/entities.py` I added the following: 

```
RctPowerSensorEntityDescription(
        get_device_info=get_inverter_device_info,
        key="grid_pll[0].f",
        name="Grid Frequency",
        state_class=STATE_CLASS_MEASUREMENT,
    ),
```
The result look like this:

[image]

weltenwort, 2021-12-21:
@k3mpaxl sorry for having let this drop off my radar

And thank you for tackling the problem and sharing the solution with us! I'll incorporate this into the code over the year-end holidays as soon as I find the time.

closed 2021-12-23 by weltenwort

weltenwort, 2021-12-23:
I released the change in [v0.9.2](https://github.com/weltenwort/home-assistant-rct-power-integration/releases/tag/v0.9.2). Let me know whether it works for you.

k3mpaxl, 2021-12-23:
> I released the change in [v0.9.1](https://github.com/weltenwort/home-assistant-rct-power-integration/releases/tag/v0.9.1). Let me know whether it works for you.

I just downloaded 0.9.2 via hacs and the grid frequency monitoring is still present as it was with my modifications in 0.9.0.

Thanks for adding this.

weltenwort, 2021-12-23:
Glad to hear that. Thanks for proposing it!

PR #174 (2021-12-23): Add sensor for grid frequency without power switch
## :memo: Summary

This adds a sensor for the grid frequency that also works for systems without a power switch as requested and implemented by @k3mpaxl.

closes #134