weltenwort/home-assistant-rct-power-integration #170: Ennergy broken on 2021.12

pluskal, 2021-12-14:
## Version of the custom_component
v0.8.2

## Configuration
Integration added in Integration UI. In 2021.11 worked as expected.

## Describe the bug
After an update to the 2021.12, the energy monitor dashboard stopped working. The energy settings show an unexpected device class.

[image]

## Debug log

This doesn't seem relevant but relates to the RCT component.
```text
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:484
Integration: Sensor (documentation, issues)
First occurred: December 13, 2021, 8:05:10 PM (32 occurrences)
Last logged: December 13, 2021, 8:05:10 PM

The unit of sensor.rct_power_storage_generator_b_energy_production_day (Wh) does not match the unit of already compiled statistics (kWh). Generation of long term statistics will be suppressed unless the unit changes back to kWh
The unit of sensor.rct_power_storage_generator_b_energy_production_month (Wh) does not match the unit of already compiled statistics (kWh). Generation of long term statistics will be suppressed unless the unit changes back to kWh
The unit of sensor.rct_power_storage_generator_b_energy_production_year (Wh) does not match the unit of already compiled statistics (kWh). Generation of long term statistics will be suppressed unless the unit changes back to kWh
The unit of sensor.rct_power_storage_generator_b_energy_production_total (Wh) does not match the unit of already compiled statistics (kWh). Generation of long term statistics will be suppressed unless the unit changes back to kWh
The unit of sensor.rct_power_storage_all_generators_energy_production_total (Wh) does not match the unit of already compiled statistics (kWh). Generation of long term statistics will be suppressed unless the unit changes back to kWh
```

weltenwort, 2021-12-14:
Thanks for reporting, I'll test with the new version.

I wish the HA team working on the energy dashboard would document the requirements for sensor to be usable on it - and the changes to those requirements.

Benjamin410, 2021-12-15:
Could it be related to this change (was hint regarding device classes in release notes) https://github.com/home-assistant/core/pull/60654?

weltenwort, 2021-12-15:
Possible, but the PR suggests the old constants are still there. So I'd merely expect a deprecation notice. But it's a good place to start the investigation, thanks.

weltenwort, 2021-12-15:
Could you help me out by pasting the "Attributes" of one of the entities that the Energy dashboard complains about?

[image]

Benjamin410, 2021-12-15:
[image]

Benjamin410, 2021-12-15:
[image]

weltenwort, 2021-12-15:
Thanks! It indeed seem to be missing the `device_class` completely.

closed 2021-12-16 by weltenwort

weltenwort, 2021-12-16:
I've [released 0.9.0](https://github.com/weltenwort/home-assistant-rct-power-integration/releases/tag/v0.9.0) with a fix. I'd appreciate any feedback you can give me about the efficacy of the fix. I hope the unit confusion in the statistics recorder resolves itself or is at least easily fixed using the "Fix it" feature of the "Statistics" table in the "Developer Tools". Again, any feedback wound be appreciated.

For anyone interested, the cause for the missing `device_class` was not the move from constants to an `Enum`, but the fact that the `SensorEntity` now overrides the base `Entity`'s `device_class` property. That caused my derived `RctPowerEntity` subclass's `device_class` to be ignored.

pluskal, 2021-12-16:
Thanks for the fix! I have tested the `0.9.0`, and it fixed the problem.

Benjamin410, 2021-12-16:
Thanks also from my side the fix works and it recovered the old data.

PR #171 (2021-12-16): Fix missing `device_class` from 2021.12.1 on
## :memo: Summary

This moves the `device_class` down the `RctPowerSensorEntity` inheritance hierarchy to adapt to the fact that the `SensorEntity` now overrides the `device_class` property of the base `Entity`.

closes #170 

## :female_detective: Notes

When the integration was installed in a Home Assistant instance for the first time before this fix but after upgrading, the recorder might have incorrectly recorded statistics using `W` and `Wh` instead of the `kW` and `kWh` expected by the Energy dashboard. In order for the entities to be selectab [... cut]