weltenwort/home-assistant-rct-power-integration #205: Unknown error occured when connecting Axpert MKS4

Technaur, 2022-04-25:
Hi,

I am trying to add my RCT Axpert MK4 inverter. But I get this error.

[image]

I am running Home Assistant OS on a Raspberry Pi 3B+ 64bit. My inverter is connected via Wi-Fi and that is the correct IP address.

Does anyone have any idea what I am missing?

Thanks.

weltenwort, 2022-04-28:
Hey @Technaur,

this integration is designed to work with the PV inverters produced by the German company [RCT Power](https://www.rct-power.com). So unless your device is merely a rebranding, I doubt this integration would work for you.

weltenwort, 2022-05-08:
Please feel free to reopen if my above assumption is incorrect.

closed 2022-05-08 by weltenwort