weltenwort/home-assistant-rct-power-integration #207: Integration not working with HA 2020-05-0

k3mpaxl, 2022-05-05:
Hi @weltenwort,

Just updated to the latest version of HA and found out that your integration won't work anymore. Didn't figure out so far what is wrong, however this issue is created to inform you about the problem.

Have a nice day,

k3mpaxl

k3mpaxl, 2022-05-05:
```

Logger: homeassistant.components.sensor
Source: custom_components/rct_power/lib/entity.py:148 
Integration: Sensor (documentation, issues) 
First occurred: 08:12:33 (2 occurrences) 
Last logged: 08:12:33

Error adding entities for domain sensor with platform rct_power
Error while setting up rct_power platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 525, in _async_add_entity
    original_device_class=entity.device_class,
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
[... 1897 lines cut ...]
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
[... cut ...]
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 372, in unit_of_measurement
    native_unit_of_measurement = self.native_unit_of_measurement
  File "/config/custom_components/rct_power/lib/entity.py", line 148, in native_unit_of_measurement
    if native_unit_of_measurement := super().native_unit_of_measurement:
RecursionError: maximum recursion depth exceeded while calling a Python object

```

danielang, 2022-05-05:
Hi I also just want to create a new issue.
Here are my findings:

With the latest release of HA they changed the temperature conversion in sensor base-component/domain. See commit: [e830032b33a892835482f314e86ceeefd6fa35b1](https://github.com/home-assistant/core/commit/e830032b33a892835482f314e86ceeefd6fa35b1) 

So now if HA is queying the `unit_of_measurement` property the base method is called which is reading the `device_class` property, which it self accesses the `unit_of_measurement`  property again -> recursion.

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 525, in _async_add_entity
    original_device_class=entity.device_class,
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
  File "/config/custom_components/rct_power/lib/entity.py", line 133, in device_class
    if self.unit_of_measurement:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 375, in unit_of_measurement
    self.device_class == DEVICE_CLASS_TEMPERATURE
```

danielang, 2022-05-05:
By removing line [133](https://github.com/weltenwort/home-assistant-rct-power-integration/blob/main/custom_components/rct_power/lib/entity.py#L133) and 134 the integration should boot again.
I dont know the impact when we dont guess the device class by unit of mesurement.

danielang, 2022-05-05:
As expected removing those lines let the integration start up. Currently I dont notice a side effect. But this needs to be investigated.

RehaagJ, 2022-05-06:
I also removed the lines, but the consequence was that values did no longer show up in the energy dashboard, and long term statistics were suppressed due to invalid units. So I added the lines back, hoping for a fix to come.

pfetzing, 2022-05-06:
I created a pull request (https://github.com/weltenwort/home-assistant-rct-power-integration/pull/209) that fixes the issue in my local testing with Home Assistant 2022.5.1.

Noiders, 2022-05-07:
After update to 2022.5.1 I lost all info except for the Battery info.
I just deleted the 2 lines mentioned. Now i get energy production info.

I'm still missing the voltage measurements for the Power Sensor.

also working except for the voltages after updating to core-2022.5.2

weltenwort, 2022-05-08:
Thanks for reporting this, everyone. And special thanks to @pfetzing, who contributed a fix in #209.

I merged it and released [v0.10.0](https://github.com/weltenwort/home-assistant-rct-power-integration/releases/tag/v0.10.0). Please let me know if anything is still broken.

closed 2022-05-08 by weltenwort