weltenwort/home-assistant-rct-power-integration #212: RCT Power Portal is not showing correct data when integration is enabled

martinwoit, 2022-05-09:
Hello! Thank you for the integration you built!

But I'm facing issues with the RCT Portal Dashboard and Reports. They are missing data and the monitors are looking odd, when the integration is enabled in HA.

Have you faced this too and if yes is there any workaround?

Best,
Martin

weltenwort, 2022-05-09:
Thanks for reporting it. I'm not using the portal myself because I isolated my inverter on its own network segment. That makes it difficult to verify for me. Maybe there are other community members here that can confirm?

Overall I wouldn't be surprised by that given the rudimentary nature of the firmware and the communication protocol. There are some things you could try to narrow it down:

1. disable the integration to check whether the portal data return to normal (I guess you've already tried that?)
2. increase the polling interval in the integration settings so the likelihood of conflicts between the portal's and home assistant's polling decreases

Does that make sense?

martinwoit, 2022-05-09:
Thank you for answering!

1. I already did and yes it went good afterwards again.
2. That I will give it a try, sounds at least reasonable. Will come back with more information later.

Thank you!

weltenwort, 2022-08-05:
I'm not sure we can do anything about this, because it's a limitation of the serial communication protocol. Please feel free to reopen if you disagree.

closed 2022-08-05 by weltenwort