weltenwort/home-assistant-rct-power-integration #228: Error when connection fails

robin13, 2022-07-17:
## Version of the custom_component

0.10.0

## Configuration

```yaml
      {
        "entry_id": "498c95f23bf3a2d06e3f193bc890138f",
        "version": 1,
        "domain": "rct_power",
        "title": "RCT Power Inverter at X.X.X.X:8899",
        "data": {
          "hostname": "X.X.X.X",
          "port": 8899,
          "entity_prefix": "RCT Power Storage"
        },
        "options": {
          "frequent_scan_interval": 10,
          "infrequent_scan_interval": 180,
          "static_scan_interval": 3600
        },
        "pref_disable_new_entities": false,
        "pref_disable_polling": false,
        "source": "user",
        "unique_id": "0065B4023192",
        "disabled_by": null
      },
```

## Describe the bug

My RCT inverter shuts down every night when the sun goes down (no battery), and the home-assistant logs are subsequently filled with these error messages. I understand that in some situations it would be an error that the inverter is not available, but perhaps the situation could rather be handled by setting the state of the device to `unavailable` rather than filling up the logs with such verbose error messages?

## Debug log

```text

2022-07-16 22:05:15 ERROR (MainThread) [custom_components.rct_power] Unexpected error fetching rct_power 0065B4023192 frequent data: [Errno 113] Connect call failed ('X.X.X.X', 8899)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 205, in _async_refresh
    self.data = await self._async_update_data()
  File "/config/custom_components/rct_power/lib/update_coordinator.py", line 46, in _async_update_data
    return await self.client.async_get_data(object_ids=self.object_ids)
  File "/config/custom_components/rct_power/lib/api.py", line 90, in async_get_data
    reader, writer = await open_connection(
  File "/usr/local/lib/python3.10/asyncio/streams.py", line 47, in open_connection
    transport, _ = await loop.create_connection(
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 1064, in create_connection
    raise exceptions[0]
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 1049, in create_connection
    sock = await self._connect_sock(
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 960, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.10/asyncio/selector_events.py", line 500, in sock_connect
    return await fut
  File "/usr/local/lib/python3.10/asyncio/selector_events.py", line 535, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
OSError: [Errno 113] Connect call failed ('X.X.X.X', 8899)
```

weltenwort, 2022-07-19:
Hi @robin13, thanks for reporting this. I understand that the connection error logging must be inconvenient in a setup without a battery. How about we add a option to the connection settings in which one can choose whether to treat an unreachable device as an error or not? Would that work for your use-case?

robin13, 2022-07-19:
Hi @weltenwort ! Small world! :-)
That would be a good compromise.
It would also be good though to set the status to unavailable, if the connection fails, as this can be used by automations to take action if the device state cannot be retrieved. Currently it just has an error, and the last successfully read values are left, so e.g any automations waiting for a value change will be not be triggered and it would be good for devices relying on the power to be able to fail safe and be turned off if it cannot be determined what the available power is.

weltenwort, 2022-07-19:
> Hi @weltenwort ! Small world!
indeed, hope you're doing well :four_leaf_clover:

> It would also be good though to set the status to unavailable
yes, that would be good in any case :+1: