weltenwort/home-assistant-rct-power-integration #237: Clean install

brainboutique, 2022-08-12:
Just moving from FHEM (integration based on https://github.com/brainboutique/rct-mqtt) to HA.
Was able to install via HACS - however I believe in the install doc two things should be mentioned.

a) Install rct_client
`apt get install python3-pip
pip install rctclient==0.0.3`

b) Restart HA after install via HACS to make the integration appear AND reload the page using Ctrl-F5.

weltenwort, 2022-08-12:
Glad to hear it worked for you in the end. HA should install the `rctclient` dependency automatically on startup. That's why HACS displays a warning that it needs to be restarted before the integration is available. Or am I misremembering it?

brainboutique, 2022-08-12:
Great tool! Hm, no, for me did not auto-install... But did crank up HA 
Supervised on oob ubuntu - did not have pip installed... Maybe that's the 
reason?
Still mabe worth mentioning in case someone runs into this...

weltenwort, 2022-08-13:
Ok, thanks for clarifying. I'll take a look at that installation scenario. :+1: Hope you don't mind if I rephrase this issue's title to represent that.

andybali, 2022-09-07:
I am order today my RCT Power Storage System. I have to wait 6 months until delivery