weltenwort/home-assistant-rct-power-integration #254: Configuration

scat703, 2022-11-17:
With regards to documentation on Github I would assume to see a configuration dialog for the mandatory configuration parameters at installation time, but there is none. There is also neither any menu anywhere to do the configuration after installation nor any config file beneath /config/custom_components/rct_power. So in the end I simply can't configure the integration in any way. obviously there must be somethung wrong with my installation.

weltenwort, 2022-11-17:
Hi @scat703, the integration should ask you for the inverter's ip address, port and name when you add the integration:

[image]

[image]

[image]

What does it show in your case?

scat703, 2022-11-20:
Hi!

My fault, I only added the integrations using HACS zu far, but not more. It's working now, thanks!

BR

closed 2022-11-20 by scat703

weltenwort, 2022-11-20:
Glad to hear that it works for you!