weltenwort/home-assistant-rct-power-integration #258: Several RCT sensor values have state class "total_increasing", but their states are negative or not strictly increasing

gutwinski, 2022-12-08:
## Version of the custom_component
V0.10.0

## Configuration

```
2022-12-07 16:16:30.037 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.rct_inverter_energy_production_day from integration rct_power has state class total_increasing, but its state is negative. Triggered by state -6559.3 with last_updated set to 2022-12-07T15:09:59.999999+00:00. Please report it to the custom integration author.
2022-12-07 16:16:30.067 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.rct_inverter_energy_production_month from integration rct_power has state class total_increasing, but its state is negative. Triggered by state -6071.2 with last_updated set to 2022-12-07T15:09:59.999999+00:00. Please report it to the custom integration author.
2022-12-07 19:16:38.010 ERROR (MainThread) [custom_components.rct_power] Timeout fetching rct_power 0065A4636875 frequent data
2022-12-07 19:31:48.002 ERROR (MainThread) [custom_components.rct_power] Timeout fetching rct_power 0065A4636875 frequent data
2022-12-07 21:20:10.235 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.rct_inverter_energy_production_day from integration rct_power has state class total_increasing, but its state is not strictly increasing. Triggered by state 821.3 (829.1) with last_updated set to 2022-12-07T20:19:18.141654+00:00. Please report it to the custom integration author.
2022-12-07 21:20:10.238 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.rct_inverter_energy_production_year from integration rct_power has state class total_increasing, but its state is not strictly increasing. Triggered by state 1849661.9 (1849669.6) with last_updated set to 2022-12-07T20:19:18.142184+00:00. Please report it to the custom integration author.
2022-12-07 21:20:10.240 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.rct_inverter_energy_production_total from integration rct_power has state class total_increasing, but its state is not strictly increasing. Triggered by state 1849661.9 (1849669.6) with last_updated set to 2022-12-07T20:19:18.142445+00:00. Please report it to the custom integration author.
2022-12-07 21:25:10.237 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.rct_inverter_energy_production_month from integration rct_power has state class total_increasing, but its state is not strictly increasing. Triggered by state 1302.6 (1310.4) with last_updated set to 2022-12-07T20:22:20.127872+00:00. Please report it to the custom integration author.
```

## Describe the bug
Several RCT sensor values w.r.t. energy production have state class "total_increasing", but their states are reported negative or not strictly increasing. I noticed these warnings yesterday when my RCT system was running a battery capacity tests, i.e. charging the battery to 100% (if needed with power from the grid) and then draining the battery to 0%. The system does that regularly about once a month.

In the screenshot, there are also other decreasing events. These probably come from times when the battery is empty (its winter after all) and needs to pull power from the grid for its internal logic.

[image]

## Debug log
n/a

weltenwort, 2022-12-08:
Thanks a lot for reporting! This sounds similar to #202.

gt026328, 2023-12-07:
Same Bug on Homeassistant 2023.12.0
2023-12-07 02:10:10.223 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.rct_dc_8_inverter_energy_production_total from integration rct_power has state class total_increasing, but its state is not strictly increasing. Triggered by state 21194758.0 (21194764.0) with last_updated set to 2023-12-07T01:06:27.126485+00:00. Please create a bug report at https://github.com/weltenwort/rct-power/issues