weltenwort/home-assistant-rct-power-integration #264: Battery calibration

OrsagFra, 2022-12-31:
**Know if there is already running battery calliation**
Know if there is already running battery callibration. If possible to know that there is phase of charging and also phase of discharging.
This can help to use energi during discharging instead of grid feeding and in case of charging there can be possibe to increase chargind during peak off period.

Thank you

weltenwort, 2022-12-31:
Hi @OrsagFra,

let me paraphrase to make sure I understand your request correctly: You're proposing to add a sensor that indicates when the calibration is running?

I've found the register `power_mng.bat_next_calib_date`, which indicates the next calibration date. There are also other registers that relate to calibration, but their semantics are not documented. It will take some guessing and reverse engineering to understand them.

Please correct me if I misunderstood your proposal.

OrsagFra, 2022-12-31:
Hello @weltenwort,
yes you are absolutely right.
Nevertheless that I can change the date and time of calibration, the system sometimes shift slightly the time and date of calibration. I also discover that there are possible 2 ways of calibration - charge to 100 % and than discharge to 0 % (usually in summer), but last month I saw that system firstly discharge to 0 % and than charge to 100 %.

patrickfe, 2023-02-27:
I came across this feature request and wonder if it would be possible to add the option to manually initiate the battery calibration process? 

As I understand it, the calibration process involves charging the battery to 100% and power can be drawn from the grid to do this. Perhaps this process can also be used to take advantage of dynamic electricity pricing. This provides the opportunity to charge the battery at times when electricity prices are lower. 

Scenario: 
In the winter months, when the battery is mostly empty anyway, a battery calibration process could start on a regular basis depending on the electricity prices.For example, the price is cheaper (e.g. 50% lower) between 4-5am. During this time the battery calibration process could start and charge the battery to 100%

After thinking about this further, I wanted to ask if there are any limitations to this approach due to conversion losses.

raserj, 2023-03-19:
> 

This feature would be great

oliverrahner, 2023-04-11:
Just for reference, this is how it looks from inside the RCT App:
[image]

oliverrahner, 2023-04-11:
After a bit of reverse engineering I see that the register this is based on is `0x70a2af4f` (clear name is `battery.bat_status`).

If `value & 1032 == 0` ==> calibration is running ["(calib.)"] (don't know why this requires two bits to match...)
If `value & 2048 == 0` ==> balance is running ["(balance)"]

OrsagFra, 2023-04-12:
Thank you, it's great.
For me as laik, I didn't find such entity in Your integration for RCT. Do I anything wrong?

SaxonWood, 2023-11-08:
I reverse engineered the OID "battery.bat_status" as well. I observed these bits (well: here denotes as decimal values) set:
                #0 =  normal mode / idle
                # 8 = calibrating - charging phase (0% --> 100%)
                # 1024 = calibrating - discharge phase (xx% --> 0%)
                # 2048 = balancing the battery units (charge to 100%)
In final balancing phase, when the inverter has reached the 99.9% SOC and tries to balance the _n_ battery units, the OID "power_mng.state" has value 2, rather than the usual value 3.
                
Info: Inverter SW version = 2.3.5661

weltenwort, 2024-01-19:
@SaxonWood thanks a lot for finding out these details. I used them in https://github.com/weltenwort/home-assistant-rct-power-integration/pull/349. Have you observed value 8 and 1024 also individually or only in combinations as reported by @oliverrahner?

closed 2024-01-19 by weltenwort

SaxonWood, 2024-01-19:
The values 8 and 1024 appeared only individually on my RCT inverter.

weltenwort, 2024-01-19:
Good to know :+1: I'll add them to the possible state values, then.

NCIceWolf, 2024-05-22:
Hello fellow rct inverter & battery users!

Since we did some rework on the rct adapter for ioBroker (https://github.com/aruttkamp/ioBroker.rct), I wanted to let you know as well:

With the latest update to 2.3.5676 und Power Battery BMS Master Software to 5607 I was able to observe the following battery-states.
1, 3 & 5 are a little "uncertain", since the documentation is a nightmare.
I saw some of that information in the RCT power app during the upgrade of the inverter and the BMS (like "CAN connection failure").
We did not see any "use-case" for these states until now, so right now, they are just for documentation.

* 0 = charge/discharge (normal operation)
* 1 = idle (no CAN-connection inverter -> battery)
* 3 = connecting (inverter -> battery)
* 5 = synchronizing (inverter -> battery)
* 8 = calibrating - charging phase (0% --> 100%)
* 1024 = calibrating - discharge phase (xx% --> 0%)
* 2048 = balancing the battery units (charge to 100%)

Cheers 🍻