weltenwort/home-assistant-rct-power-integration #278: Link to documentation is incorrect

mqatgit, 2023-02-09:
Clicking on the "Documentation" link (and also the ? symbols in pop up windows) leads here: https://github.com/weltenwort/rct-power.

But there is nothing to be found there...

[image]

Observed with current version 0.10.0.

Noiders, 2023-02-09:
manifest.json line 3 and 4 need to be updated with correct url

  "documentation": "https://github.com/weltenwort/home-assistant-rct-power-integration",
  "issue_tracker": "https://github.com/weltenwort/home-assistant-rct-power-integration/issues",

weltenwort, 2023-02-10:
Thank you for reporting this. The fix from #280 should be included in the next release.

PR #280 (2023-02-10): Fix documentation and issue tracker links in the manifest
## :memo: Summary

This fixes the links to the documentation and issue tracker in the component manifest file.

closes #278
closed 2023-02-10 by weltenwort