weltenwort/home-assistant-rct-power-integration #279: Implausible values from Grid Energy Consumption Total

rhoegen, 2023-02-10:
Hi all,

first of all thanks a lot for the great integration. It is really working well and is used as the heart of my energy management with HA.

I currently have an issue that probably has its origin in an update from RCT. I have two PowerStorage DC 10.0 in Master/Slave Mode. The Master always provided the "Grid Energy Consumption Total" Entity without a problem. After the Update to the control software "2.3.5658" (Battery Software 5597), the Entity updates only daily with always exactly the same value, which seems implausible. 

[image]
[image]

What I've tried to do:

- Restart RCT devices
- Restart HA
- Reload Integration
- Remove Entity from Energy Dashboard, Restart HA and add Entity again

Does anyone have a similar issue with the current software from RCT?

Thanks and best regards

René

rhoegen, 2023-02-10:
Update: Switched the Grid Energy Consumption with the Entity from the Slave and now it looks more realistic:
[image]

I will check this with the energy meter from the provider to see if the overall numbers match.

weltenwort, 2023-02-10:
@rhoegen, thanks for reaching out! If the inverter is responding with implausible values there's unfortunately little we can do. In https://github.com/svalouch/python-rctclient/issues/22 the community describes various manifestations of similar symptoms. Please let us know if you find out anything that can (in)validate the assumption that it's a problem in the inverter's firmware.