weltenwort/home-assistant-rct-power-integration #311: No battery status after firmware upgrade

robin13, 2023-05-12:
Since upgrading RCT Power Storage 10.0 (PS 10.0 B59G) to latest firmware 2.3.5661 there are no values for the RCT Power battery device...

weltenwort, 2023-05-12:
hm, I can't reproduce that. It seems to work for me on that same version. By "no values" you mean the entities are gone? Or they just don't receive new values?

robin13, 2023-05-12:
All the entities under the device storage battery are unavailable.
I don't see anything related in the logs... Is there something I can do to get more insight into what is missing?

weltenwort, 2023-05-28:
Sorry for the delay in answering :see_no_evil: You could try to increase the log level to `debug` for the integration by setting this in your Home Assistant `configuration.yaml`:

```yaml
logger:
  default: info
  logs:
    custom_components.rct_power: debug
```

robin13, 2023-06-15:
I tried enabling the debugging, but did not see anything there that helped me understand. The only slightly off thing I noticed was:
```
2023-06-11 19:19:26.098 DEBUG (MainThread) [custom_components.rct_power.lib] Decoded data for object 959930bf (battery.soc): 0.5
2023-06-11 19:19:26.286 DEBUG (MainThread) [custom_components.rct_power.lib] Decoded data for object 959930bf (battery.soc): 0.9671242237091064
```
This is strange that it seems to be reading both an invalid SOC (first line) and an invalid one...

In any case... I deleted the devices and created them new. With the newly created ones, the battery state is again visible. It would be great if you can identify something in the logs to avoid this happening again, as it was quite a bother to change the entity ids in all the automations etc. that refer to these.

[Uploading rct_power_debug.log.gz…]()

weltenwort, 2023-06-15:
Interesting, it almost looks like the inverter responded to the query with two different values :thinking:

Thanks for enabling the debug logging. Unfortunately the upload didn't quite work, apparently. :grimacing: