weltenwort/home-assistant-rct-power-integration #314: Change Request for all negativ values into positives.

Sectorchan, 2023-06-06:
For what reasons are the values on the entities are negativ?
From my point of view, the values of the entitiy should always are positiv, even if they are consumed.
Later on, during calculation in any other application, home assistant or Grafana the values of the entities should be subsctracted from another value, but in the case the value are already negative and you can not subsctract them.

My request would be to change all negative values into positiv.

brlnr23, 2024-09-08:
imho, positive/negative values are pretty fine. One could use *one* entity for automations etc. Changing all to positive would either cause two entities to be needed for everything (such as `battery_power_charge` and `battery_power_discharge`) or even more complicated: one entity for the value (`battery_power`) and a "type" entity (`battery_charging` or `battery_discharging`). 

I am using Grafana also quite extensively. May I ask what your use case is exactly and what database you use behind the scenes (influx, timescaledb, ...)? Maybe there's a better solution for that?

Sectorchan, 2024-09-26:
I use grafana, or better want to use...

Lets use `sensor.rct_power_consumer_energy_consumption_total` It has a value of `-61.848,6 Wh`
Per definition a consumption is always that I take away and a negativ value, would mean imo that it is again +
Also its my opinion that this negativ value should be done through the calculation with the proper "-" operator.
Same as` sensor.rct_power_grid_energy_production_total`

Your sample with the battery_power seems for me also reasonable and would take much programming to check if battery_charging is true or false.