weltenwort/home-assistant-rct-power-integration #322: Always "Initalizing" & "Failed setup. will retry."

andyrockt, 2023-07-18:
I have issues with the integration with my power rtc. with the app it's possible to get the data.
can't find out whats the problem.
If needed I send you more debug logs.
Logs: ( i've removed my ip)
```
This error originated from a custom integration.

Logger: custom_components.rct_power
Source: helpers/update_coordinator.py:262
Integration: RCT Power (documentation, issues)
First occurred: July 17, 2023 at 12:59:16 PM (1260 occurrences)
Last logged: 7:46:06 PM

Timeout fetching rct_power 0065A4630378 frequent data
Timeout fetching rct_power 0065A4630378 infrequent data
Timeout fetching rct_power 0065A4630378 static data

This error originated from a custom integration.

Logger: custom_components.rct_power
Source: custom_components/rct_power/lib/api.py:90
Integration: RCT Power (documentation, issues)
First occurred: July 17, 2023 at 9:19:14 PM (1036 occurrences)
Last logged: 7:41:37 PM

Unexpected error fetching rct_power 0065A4630378 frequent data: [Errno 113] Connect call failed ('my ip', 8899)
Unexpected error fetching rct_power 0065A4630378 infrequent data: [Errno 113] Connect call failed ('my ip', 8899)
Unexpected error fetching rct_power 0065A4630378 static data: [Errno 113] Connect call failed ('my ip', 8899)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 283, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/rct_power/lib/update_coordinator.py", line 46, in _async_update_data
    return await self.client.async_get_data(object_ids=self.object_ids)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/rct_power/lib/api.py", line 90, in async_get_data
    reader, writer = await open_connection(
                     ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/streams.py", line 48, in open_connection
    transport, _ = await loop.create_connection(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1085, in create_connection
    raise exceptions[0]
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1069, in create_connection
    sock = await self._connect_sock(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 973, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 634, in sock_connect
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 674, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
OSError: [Errno 113] Connect call failed ('my ip', 8899)
```
Thanks for a tipp or your help!
Andy

weltenwort, 2023-08-07:
Hey @andyrockt, the timeout indicates that Home Assistant can't reach the inverter using the ip address and port that you configured it for. If you are running Home Assistant on a different subnet than the inverter there must be a valid route from the former to the latter.

Hope this helps somewhat?

andyrockt, 2023-08-10:
Hi @weltenwort 
No, it's running in the same subnet. so perhaps it's something with the wifi or the power rct.. could something be wrong configured with the power rct? (port or something line this)

weltenwort, 2023-08-11:
That's curious :thinking: Two things to check:

* Which inverter model do you have? Maybe there's a different model that I haven't seen yet.
* Can you ping the RCT inverter from the machine that runs Home Assistant? If you're using a supervised installation you could use the "Terminal & SSH" addon for that.

andyrockt, 2023-08-13:
Should be the "RCT Power Inverter 6.0".

Yes, via the Terminal from the Home Asisstant Installation ping is possible. The pings are between 3 - 5ms, but with no interrupt.