weltenwort/home-assistant-rct-power-integration #325: time to update custom_components.rct_power.lib.entity.RctPowerSensorEntity

Sectorchan, 2023-10-08:
I opened the logfile and saw this:
2023-10-08 18:34:23.451 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.rct_power_consumer_power (<class 'custom_components.rct_power.lib.entity.RctPowerSensorEntity'>) took 0.847 seconds. Please create a bug report at https://github.com/weltenwort/rct-power/issues

Not sure what informations I should provide. Sorry

weltenwort, 2024-01-07:
Hey @Sectorchan, thanks for reporting this. I've seen these warnings to. They are just a mechanism of Home Assistant to nudge integration developers to pay attention to performance.

Unfortunately the proprietary API that the inverter implements is difficult to use efficiently. I have a few ideas what could be improved (such as the scheduling of state updates), but it exceeds my available time right now. Feel free to propose a contribution, though. :innocent:

closed 2024-01-07 by weltenwort

Sectorchan, 2024-01-07:
Unfortunately this is out of the scope of my programming skills :( 

I would be happy if I could help