weltenwort/home-assistant-rct-power-integration #331: Not working in 2024.1.0

705Marc, 2024-01-03:
I just updated to the new Release of HomeAssistant,

the integration is failing to load.

Logger: homeassistant.loader
Source: loader.py:822
First occurred: 21:11:05 (1 occurrences)
Last logged: 21:11:05

Unexpected exception importing component custom_components.rct_power
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 822, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/rct_power/__init__.py", line 25, in <module>
    from .lib.entities import all_entity_descriptions
  File "/config/custom_components/rct_power/lib/entities.py", line 7, in <module>
    from .device_info_helpers import get_battery_device_info, get_inverter_device_info
  File "/config/custom_components/rct_power/lib/device_info_helpers.py", line 4, in <module>
    from .entity import RctPowerEntity
  File "/config/custom_components/rct_power/lib/entity.py", line 201, in <module>
    @dataclass
     ^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1230, in dataclass
    return wrap(cls)
           ^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1220, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1027, in _process_class
    _init_fn(all_init_fields,
  File "/usr/local/lib/python3.11/dataclasses.py", line 545, in _init_fn
    raise TypeError(f'non-default argument {f.name!r} '
TypeError: non-default argument 'object_infos' follows default argument

blueslow, 2024-01-04:
705Marc info is more complete but adding  my rudimentary log.

Setup failed for custom integration 'rct_power': Unable to import component: Exception importing custom_components.rct_power
02:10:54 – (FEL) setup.py
Unexpected exception importing component custom_components.rct_power
02:10:54 – (FEL) loader.py

POWER_VOLT_AMPERE was used from rct_power, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfApparentPower.VOLT_AMPERE instead, please create a bug report at https://github.com/weltenwort/rct-power/issues

janrothkegel, 2024-01-04:
Any homebrewn workaround so far? :-)

chrjsorg, 2024-01-04:
I did a downgrade of HASS because of this.

If you use Docker Compose just pin the version in the docker-compose.yml

```
homeassistant:
    container_name: homeassistant
    #image: "ghcr.io/home-assistant/home-assistant:stable" stable is 2024.1..
    image: "ghcr.io/home-assistant/home-assistant:2023.12.4"
    volumes:

docker-compose pull 
docker-compose up -d
```

kantokant, 2024-01-04:
Same here, productive HAOS on RPI 3B, running
Core 2024.1.0
Supervisor 2023.12.0
Operating System 11.2
Frontend 20240103.3

After update to Core 2024.1.0 the RCT integration is not loading anymore, throwing these messages in error logs:

```
Logger: homeassistant.setup
Source: setup.py:251
First occurred: 08:37:38 (1 occurrences)
Last logged: 08:37:38

Setup failed for custom integration 'rct_power': Unable to import component: Exception importing custom_components.rct_power
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 822, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/rct_power/__init__.py", line 25, in <module>
    from .lib.entities import all_entity_descriptions
  File "/config/custom_components/rct_power/lib/entities.py", line 7, in <module>
    from .device_info_helpers import get_battery_device_info, get_inverter_device_info
  File "/config/custom_components/rct_power/lib/device_info_helpers.py", line 4, in <module>
    from .entity import RctPowerEntity
  File "/config/custom_components/rct_power/lib/entity.py", line 201, in <module>
    @dataclass
     ^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1230, in dataclass
    return wrap(cls)
           ^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1220, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1027, in _process_class
    _init_fn(all_init_fields,
  File "/usr/local/lib/python3.11/dataclasses.py", line 545, in _init_fn
    raise TypeError(f'non-default argument {f.name!r} '
TypeError: non-default argument 'object_infos' follows default argument

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 251, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 830, in get_component
    raise ImportError(f"Exception importing {self.pkg_path}") from err
ImportError: Exception importing custom_components.rct_power
```

```
Logger: homeassistant.loader
Source: loader.py:822
First occurred: 08:37:38 (1 occurrences)
Last logged: 08:37:38

Unexpected exception importing component custom_components.rct_power
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 822, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
[... 19 lines cut ...]
  File "/usr/local/lib/python3.11/dataclasses.py", line 1230, in dataclass
    return wrap(cls)
           ^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1220, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1027, in _process_class
    _init_fn(all_init_fields,
  File "/usr/local/lib/python3.11/dataclasses.py", line 545, in _init_fn
    raise TypeError(f'non-default argument {f.name!r} '
TypeError: non-default argument 'object_infos' follows default argument
```

```
Logger: homeassistant.const
Source: helpers/deprecation.py:204
First occurred: 08:37:37 (12 occurrences)
Last logged: 08:37:37

POWER_VOLT_AMPERE was used from rct_power, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfApparentPower.VOLT_AMPERE instead, please create a bug report at https://github.com/weltenwort/rct-power/issues
POWER_WATT was used from rct_power, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfPower.WATT instead, please create a bug report at https://github.com/weltenwort/rct-power/issues
TEMP_CELSIUS was used from rct_power, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfTemperature.CELSIUS instead, please create a bug report at https://github.com/weltenwort/rct-power/issues
TEMP_FAHRENHEIT was used from rct_power, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfTemperature.FAHRENHEIT instead, please create a bug report at https://github.com/weltenwort/rct-power/issues
TEMP_KELVIN was used from rct_power, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfTemperature.KELVIN instead, please create a bug report at https://github.com/weltenwort/rct-power/issues
```

heikone, 2024-01-04:
Maybe my log also helps - appeared after update to Homeasistant Core 2024.1:

My system:

```
Core 2024.1.0
Supervisor 2023.12.0
Frontend 20240103.3
```

My log:

```
Logger: homeassistant.setup
Source: setup.py:251
First occurred: 09:15:23 (1 occurrences)
Last logged: 09:15:23

Setup failed for custom integration 'rct_power': Unable to import component: Exception importing custom_components.rct_power
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 822, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/rct_power/__init__.py", line 25, in <module>
    from .lib.entities import all_entity_descriptions
  File "/config/custom_components/rct_power/lib/entities.py", line 7, in <module>
    from .device_info_helpers import get_battery_device_info, get_inverter_device_info
  File "/config/custom_components/rct_power/lib/device_info_helpers.py", line 4, in <module>
    from .entity import RctPowerEntity
  File "/config/custom_components/rct_power/lib/entity.py", line 201, in <module>
    @dataclass
     ^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1230, in dataclass
    return wrap(cls)
           ^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1220, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1027, in _process_class
    _init_fn(all_init_fields,
  File "/usr/local/lib/python3.11/dataclasses.py", line 545, in _init_fn
    raise TypeError(f'non-default argument {f.name!r} '
TypeError: non-default argument 'object_infos' follows default argument

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 251, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 830, in get_component
    raise ImportError(f"Exception importing {self.pkg_path}") from err
ImportError: Exception importing custom_components.rct_power

```

janrothkegel, 2024-01-04:
I fixed the issue for now. Feel free to use my fork https://github.com/janrothkegel/home-assistant-rct-power-integration/tree/main which also contains changes to display calibration and balancing as entities from @oliverrahner

chrjsorg, 2024-01-04:
@janrothkegel Just applied the changes and updated HASS again to 2024.*. 
It works, thanks.

diomedon, 2024-01-04:
@janrothkegel @chrjsorg I am not a HASS expert. Might you quickly explain, how I can add this fork? I probably have to reconfigure the energy dashboard :-(

janrothkegel, 2024-01-04:
@diomedon You might want to follow this explanation: https://hacs.xyz/docs/faq/custom_repositories/

After adding the repository, you can open and download it via HACS. However, you will also need to remove the old device and install it with the same name (then the entity names for your energy dashboard should not change).

Other option: Manually open /homeassistant/custom_components/rct_power/lib/entity.py and replace with the content from my repository.

ChristophCaina, 2024-01-04:
> 705Marc info is more complete but adding my rudimentary log.
>
> Setup failed for custom integration 'rct_power': Unable to import component: Exception importing custom_components.rct_power 02:10:54 – [...]

if this is the only reason, the integration is failing, then this should probably be an easy fix.

It seems mostly be related with the file `/custom_components/rct_power/lib/device_class_helpers.py`
Which is using depricated constants:

```
from homeassistant.const import (
    ELECTRIC_CURRENT_AMPERE,
    ELECTRIC_CURRENT_MILLIAMPERE,
    ELECTRIC_POTENTIAL_MILLIVOLT,
    ELECTRIC_POTENTIAL_VOLT,
    ENERGY_KILO_WATT_HOUR,
    ENERGY_WATT_HOUR,
    POWER_KILO_WATT,
    POWER_VOLT_AMPERE,
    POWER_WATT,
    TEMP_CELSIUS,
    TEMP_FAHRENHEIT,
    TEMP_KELVIN,
)
```

OrsagFra, 2024-01-04:
I tried alternative option, proposed by [janrothkegel](https://github.com/janrothkegel), but it didn't work. Standard way, by remove old and add new integration from mentioned repository work well.
Thank you for help.

OrsagFra, 2024-01-04:
I tried alternative option, proposed by [janrothkegel](https://github.com/janrothkegel), but it didn't work. Standard way, which he proposed as first, which means remove old and add new integration from repository work well.
Thank you for help.

janrothkegel, 2024-01-05:
> if this is the only reason, the integration is failing, then this should probably be an easy fix.
>
> It seems mostly be related with the file `/custom_components/rct_power/lib/device_class_helpers.py` [...]

No, this is just a warning. It needs to be fixed until the due date.

The actual cause for the issue that we discuss here is probably a [change of the underlying SensorEntityDescription and BinarySensorEntityDescription](https://github.com/home-assistant/core/commit/104bcc64b7da3195c24ed58f1ec60dff6195ee8d) in 2024.1, which screwed up the multiple inheritance hierarchy in the RCT Power integration. However, I really have no clue on Python ^^

@weltenwort Is there a chance to merge the various pull requests into the integration? Can I somehow help doing this?

weltenwort, 2024-01-05:
Hey folks, sorry for the silence. I'll look into merging PRs and fixes over the weekend. Thanks for reporting and investigating!

wora55, 2024-01-05:
Meine Version läuft auch wieder. Ich habe nur die beiden Python-Files (state_helpers.py und entity.py) in der bestehenden Version aktualisiert (über ssh) und neu gestartet. Reicht auch.

DukeMini, 2024-01-05:
Dito. Beide Files aus Jans Fork übernommen und es läuft wieder.

zacatecnik, 2024-01-05:
Hi, I want to apologize in advance. I am a newbie amateur, running HA by trial and error. So I am asking for advice from experienced people.
I have the same integration problem after the first HA update in 2024.
Please do I understand correctly that I just need to overwrite my entity.py file (bugfixes) ? It doesn't work :-( Thank you for any detailed help.

wora55, 2024-01-07:
No, you need to copy two files:  state_helpers.py and entity.py (and restart your HA)
Or just wait, there is a new version of the integration supposed to be available after the weekend. 

Wolfgang

zacatecnik schrieb am Freitag, 5. Januar 2024 22:33:07 (+01:00):

Hi, I want to apologize in advance. I am a newbie amateur, running HA by trial and error. So I am asking for advice from experienced people.
I have the same integration problem after the first HA update in 2024.
Please do I understand correctly that I just need to overwrite my entity.py file (bugfixes) ? It doesn't work :-( Thank you for any detailed help.

—

Wolfgang Radke
Schubertstr. 6f
76344 Eggenstein

closed 2024-01-07 by weltenwort

blueslow, 2024-01-07:
Great and good  work Felix.

zacatecnik, 2024-01-07:
> […](#)
> -- Wolfgang Radke Schubertstr. 6f 76344 Eggenstein

Okay, thank you. I did, and it's reporting another error.
I've seen that there are already more currently modified files on github. Should I copy all the current ones ?

weltenwort, 2024-01-07:
Hi folks, I just published https://github.com/weltenwort/home-assistant-rct-power-integration/releases/tag/v0.11.0.

Due to the way the metaclass/dataclasses change was made in HA, it was difficult to release a version that works in both HA 2023.x and 2024.x. So I set the minimum required version to 2024.1.0, which means the update order is first HA core and then this integration.

Please let me know if anything breaks unexpectedly.

kantokant, 2024-01-07:
works like a charm, on standard update path from core 2023.12.4 to 2024.1.2 and rct integration update via hacs standard functionality
thank so you much, great work! 👍

diomedon, 2024-01-07:
I just installed the new version. Works. Thanks a lot also from my side.

gt026328, 2024-01-07:
Just updated HASS from 2023.12.4 to 2024.1.2 and then opened HACS. There was already the information about an update from rct-power. Just pressed update and all worked fine. Restarted HASS-Core and all is running fine.
Thank you very much for your effort and great job done! Thanks a lot

PR #335 (2024-01-07): Restore compatibility with 2024.1.0
## :memo: Summary

This changes the entity description dataclasses to be compatible with the changes made in https://github.com/home-assistant/core/pull/105211. It intentionally doesn't pull in any other changes so that users can choose to only install the fix. New features will be released in a follow-up version.

closes #331