weltenwort/home-assistant-rct-power-integration #338: Felix kommt eine korregierte Version?

knieshe1, 2024-01-07:
Bin Neuling
leider nicht so fit im programmieren
kommt eine neue Version von RCT-Power welche mit Version HA 2024.1.2 läuft.

Danke im Voraus
Heinz

weltenwort, 2024-01-07:
Hallo Heinz, habe gerade Version 0.11.0 released, die mit 2024.1.2 und aufwärts funktionieren sollte. Bitte lass es mich wissen wenn es zu Problemen führt.

knieshe1, 2024-01-07:
> <weltenwort/home-assistant-rct-power-integration/issues/338/1880143818@
> github.com>

Kannst du mir bitte erklären wie ich diese Version in mein System bekomme.
Ich habe HA als VM in PROXMOX auf einem Futron Q920 installiert.

Schonmal vielen im voraus

LG aus dem Südwest
Heinz

weltenwort, 2024-01-07:
Wenn Du die Integration über HACS installiert hast, dann müsste es Dir ein Update anbieten:

[image]

[image]

knieshe1, 2024-01-07:
Recht schönen guten Abend Felix,

dein Fix hat super funktioniert😁

Von "Hand" habe ich es nicht hinbekommen.
Dann habe ich die Update-Info gesehen.
Das Einspielen ist problemlos durchgelaufen.
Alle Daten sind korrekt wiederhergestellt.

Nochmals recht herzlichen Dank und Gruß
aus der südwestlichsten Ecke des Landes

Heinz (70+)

weltenwort, 2024-01-07:
Freut mich zu hören :tada: Viel Erfolg weiterhin!