weltenwort/home-assistant-rct-power-integration #376: Control the system (actually set battery charge level dynamically)

heikone, 2024-03-10:
Hello together,

just a thing, that came to my mind:
Is there a possibility to change the charge level of the battery beside setting it in the RCT app?
My goal would be to control the charge level dynamically from Homeassistant.

I believe this is an enhancement and maybe somebody give any hint how to manage this.

Greetings,
Heiko

rhoegen, 2024-03-14:
I would also appreciate this feature, since the internal SoC feature of the RCT devices is not as flexible as I would want it to be. It would be sufficient to change the "non-island" Target SoC, which is available in the app through the installer menu.

hanzsez, 2024-03-15:
+1 to this, even it would mean that the devs have to implement the write function (which they do not want to, for obvious reasons). But having just battery soc manipulation as write would benefit me a lot as well

philoxio, 2024-04-07:
This would be a huge benefit. Specially if you want to charge an EV and "hold" the SoC of the battery, instead of having to wait for the battery to be fully charged.

heikone, 2024-04-23:
@philoxio, this exactly my case of application!

kaitimmer, 2024-05-07:
I agree, being able to control the SoC Target and maybe max SoC would be awesome.

do-gooder, 2024-08-02:
```
#!/usr/bin/env python3

## power_mng.soc_strategy		0: SOC-Ziel = SOC, 1: Konstant, 2: Extern, 3: Mittlere Batteriespannung, 4: Intern, 5: Zeitplan (default: 4)
## power_mng.soc_target_set		Force SOC target (default: 0.5) --> (1: Konstant)
## power_mng.battery_power_extern	Battery target power (positive = discharge) --> (2. Extern)

## power_mng.soc_min			Min SOC target (default 0.07) --> ggf. als Ersatzstrom-Reserve
## power_mng.soc_charge_power		Charging to power_mng.soc_min with given W (default: 100)
## power_mng.soc_charge			Trigger for charing to soc_min (default: 0.05)

import sys
import socket
import select
from rctclient.frame import make_frame, ReceiveFrame
from rctclient.registry import REGISTRY
from rctclient.types import Command
from rctclient.utils import decode_value, encode_value

def show_help():
    print("Usage:")
    print("  test.py get <parameter> --host=<ip_address_or_hostname>")
    print("  test.py set <parameter> <value> --host=<ip_address_or_hostname>")
    print("\nValid Parameters:")
    print("  power_mng.soc_strategy - SOC charging strategy")
    print("    Valid Values:")
    print("      0: SOC target = SOC (State of Charge)")
    print("      1: Konstant (Constant)")
    print("      2: Extern (External)")
    print("      3: Mittlere Batteriespannung (Average Battery Voltage)")
    print("      4: Intern (Internal)")
    print("      5: Zeitplan (Schedule)")
    print("    Default Value: 4 (Internal)")
    print("  power_mng.soc_target_set - Force SOC target")
    print("    Valid Range: 0.00 to 1.00, with at most two decimal places")
    print("    Default Value: 0.50")
    print("  power_mng.battery_power_extern - Battery target power")
    print("    Valid Range: -6000 to 6000")
    print("      Positive values indicate discharge, negative values indicate charge")
    print("    Default Value: 0")
    print("  power_mng.soc_min - Min SOC target")
    print("    Valid Range: 0.00 to 1.00, with at most two decimal places")
    print("    Default Value: 0.07")
    print("  power_mng.soc_charge_power - Charging power to reach SOC target")
    print("    Default Value: 100")
    print("  power_mng.soc_charge - Trigger for charging to SOC_min")
    print("    Default Value: 0.05")

def set_value(parameter, value, host):
    valid_parameters = [
        "power_mng.soc_strategy",
        "power_mng.soc_target_set",
        "power_mng.battery_power_extern",
        "power_mng.soc_min",
        "power_mng.soc_charge_power",
        "power_mng.soc_charge"
    ]

    if parameter not in valid_parameters:
        print(f"Error: Invalid parameter '{parameter}'.")
        show_help()
        sys.exit(1)

    if parameter == "power_mng.soc_strategy" and value not in ["0", "1", "2", "3", "4", "5"]:
        print(f"Error: Invalid value '{value}' for parameter '{parameter}'.")
        show_help()
        sys.exit(1)
    elif parameter == "power_mng.soc_strategy":
        try:
            value = int(value)
[... 117 lines cut ...]
            host = arg[len("--host="):]
            break

    if not host:
        print("Error: Host parameter is missing.")
        show_help()
        sys.exit(1)

    if subcommand == "set":
        if len(sys.argv) != 5:
            print("Error: Please provide a parameter, a value, and a host to set.")
            show_help()
            sys.exit(1)
        set_value(sys.argv[2], sys.argv[3], host)
    elif subcommand == "get":
        if len(sys.argv) != 4:
            print("Error: Please provide a parameter and a host to get.")
            show_help()
            sys.exit(1)
        get_value(sys.argv[2], host)
    else:
        print(f"Error: Unknown command '{subcommand}'.")
        show_help()
        sys.exit(1)
```

heikone, 2024-08-02:
Thanks a lot for this proposal!
I've tried your script to GET values from my system and it works.

But I'm confused a little bit, because I've limited maximum SOC to 80% (with strategy "constant") at the moment and I guess, the proper method is `power_mng.soc_target_set` to get this.
But I was wrong. `python test.py get power_mng.soc_target_set --host=192.168.0.99` is reporting 1.0 instead of expected 0.8.

Maybe you advice some help!

Greetings, Heiko

do-gooder, 2024-08-02:
Unfortunately there is no documentation. We have to find out for ourselves how the settings work.

[Special RCT Infos](https://www.photovoltaikforum.com/thread/159603-rct-power-storage-soc-zielauswahl-extern-nutzen/?postID=2656687#post2656687)

[Starting on Page 32](https://www.rct-power.com/de/download-bereich-de.html?file=files/Download-Bereich/Download%20Bereich%20EN/3.1_RCT%20Power%20Storage%20DC%208-10/RCT-Power-Storage-DC10_Manual_Web24V1EN.pdf)

Please share your insights!

heikone, 2024-08-02:
The first one is helpful, I think.
@do-gooder, did you test to use `power_mng.soc_max` instead of `power_mng.soc_target_set` to dynamically limit the charge?
This register is listed in [rctclient documentation](https://rctclient.readthedocs.io/en/latest/inverter_registry.html#power-mng) and also [#8](https://www.photovoltaikforum.com/thread/159603-rct-power-storage-soc-zielauswahl-extern-nutzen/?postID=2864743#post2864743) and [#12](https://www.photovoltaikforum.com/thread/159603-rct-power-storage-soc-zielauswahl-extern-nutzen/?postID=2867630#post2867630) in thread "RCT Power Storage - SOC Zielauswahl "Extern" nutzen" you provided.

do-gooder, 2024-08-02:
My script is more than months old. But I don't use it due to lack of time.

do-gooder, 2024-08-03:
Once we have figured out how this all works, we could write a wrapper that provides the functions such as charging from the grid for Tibber, changing the SOC, etc. via an HTTP API. This is what RCT Power should have done.

heikone, 2024-08-03:
Hi, meanwhile I've tested to change 'power_mng.soc_max' by adding this to your script in the same way 'power_mng.soc_min' was already available. 
**Result: Works as expected!**
Inverter is is immediately accepting new maximum SOC level and further charges or stops, depending on the actual SOC. 

I agree, this could be the basis for any other solution. Just changing the maximum SOC seems to be of limited and acceptable risk for me at the moment. We should remember, that writing settings is not recommended by the rctclient documentation ...

When I can find time during my vacation, I will propose a solution for adjusting maximum SOC from input number of Homeassistant.

Greetings, Heiko

do-gooder, 2024-08-03:
Ok, please use [rct_writesupport](https://github.com/do-gooder/rctpower_writesupport) for your updates!