weltenwort/home-assistant-rct-power-integration #379: new IP address

1977michael, 2024-03-26:
Hello, I have set up my network again and the RCT-WR has a new IP address.
How can I change this in the integration?

dahadd, 2024-03-26:
add new devise
[image]

1977michael, 2024-03-26:
Sind dann die Entitäten neu bezeichnet? Dann würden ja die von mir angelegten Helfer nicht mehr funktionieren Von meinem iPhone gesendet

IMG_20240326_220118.jpg (view on web)

dahadd, 2024-03-26:
Andere Möglichkeit wäre die Integration komplett neu anzulegen. Die alte Konfiguration löschen...Die "neuen" Entitäten haben dann den gleichen Namen...

1977michael, 2024-03-27:
Perfekt, hat super funktioniert Danke auch für dein Engagement in Bezug auf die Integration .Von meinem iPhone gesendet

weltenwort, 2025-02-16:
Sounds like you were able to fix your issue. 🎉 Feel free to re-open if that isn't the case.

closed 2025-02-16 by weltenwort