weltenwort/home-assistant-rct-power-integration #391: Detected that custom integration 'rct_power' calls `async_add_job`, which is deprecated and will be removed in Home Assistant 2025.4

pipip, 2024-05-13:
Home Assistant Log warns about the following.
Still got some time until 2025 :)

`Detected that custom integration 'rct_power' calls async_add_job, which is deprecated and will be removed in Home Assistant 2025.4; Please review https://developers.home-assistant.io/blog/2024/03/13/deprecate_add_run_job for replacement options at custom_components/rct_power/__init__.py, line 122: hass.async_add_job( # type: ignore, please create a bug report at https://github.com/weltenwort/home-assistant-rct-power-integration/issues`

gt026328, 2024-07-05:
some more Infos from the Log-File and HA Version 2024.7
2024-07-04 12:02:13.884 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'rct_power' calls `async_add_job`, which is deprecated and will be removed in Home Assistant 2025.4; Please review https://developers.home-assistant.io/blog/2024/03/13/deprecate_add_run_job for replacement options at custom_components/rct_power/__init__.py, line 122: hass.async_add_job(  # type: ignore, please create a bug report at https://github.com/weltenwort/home-assistant-rct-power-integration/issues
2024-07-04 12:02:13.886 WARNING (MainThread) [homeassistant.helpers.frame] Detected code that calls async_forward_entry_setup for integration, rct_power with title: RCT Power Inverter at 192.168.1.163:8899 and entry_id: f005bfb9f39aa95771909c356c8f3c92, which is deprecated and will stop working in Home Assistant 2025.6, await async_forward_entry_setups instead. Please report this issue.
Stack (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2099, in async_forward_entry_setup
    report(
  File "/usr/src/homeassistant/homeassistant/helpers/frame.py", line 151, in report
    _LOGGER.warning(msg, stack_info=True)

weltenwort, 2025-02-16:
✅ This should have been fixed in [v0.14.1](https://github.com/weltenwort/home-assistant-rct-power-integration/releases/tag/v0.14.1).

💟 Thanks a lot for reporting!

closed 2025-02-16 by weltenwort