weltenwort/home-assistant-rct-power-integration #392: No longer connected to the RCT - Unexpected error fetching rct_power 0065A4639173 frequent data

andreh0, 2024-05-21:
## Version of the custom_component
RCT 0.13.1
core_2024.5.3

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

No more connection to RCT
There has been no connection for a few days now.
The RCT app is working.
## Debug log

```text

Add your logs here.
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.rct_power
Quelle: helpers/update_coordinator.py:315
Integration: RCT Power (Dokumentation, Probleme)
Erstmals aufgetreten: 19. Mai 2024 um 20:35:21 (4995 Vorkommnisse)
Zuletzt protokolliert: 10:00:10

Unexpected error fetching rct_power 0065A4639173 frequent data
Unexpected error fetching rct_power 0065A4639173 infrequent data
Unexpected error fetching rct_power 0065A4639173 static data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/rct_power/lib/update_coordinator.py", line 46, in _async_update_data
    return await self.client.async_get_data(object_ids=self.object_ids)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/rct_power/lib/api.py", line 90, in async_get_data
    reader, writer = await open_connection(
                     ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 48, in open_connection
    transport, _ = await loop.create_connection(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1078, in create_connection
    infos = await self._ensure_resolved(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1454, in _ensure_resolved
    return await loop.getaddrinfo(host, port, family=family, type=type,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 899, in getaddrinfo
    return await self.run_in_executor(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/socket.py", line 963, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
socket.gaierror: [Errno -2] Name does not resolve
```

andreh0, 2024-05-22:
Today it works again. Seven days are missing. Nothing has been changed, crazy.
But there is a log: 

Logger: homeassistant.components.sensor.recorder
Quelle: components/sensor/recorder.py:302
Integration: Sensor (Dokumentation, Probleme)
Erstmals aufgetreten: 13:15:10 (4 Vorkommnisse)
Zuletzt protokolliert: 13:15:10

Entity sensor.rct_power_storage_consumer_energy_consumption_day from integration rct_power has state class total_increasing, but its state is not strictly increasing. Triggered by state 4057.1 (4069.1) with last_updated set to 2024-05-22T11:14:35.393062+00:00. Please create a bug report at https://github.com/weltenwort/home-assistant-rct-power-integration/issues
Entity sensor.rct_power_storage_consumer_energy_consumption_month from integration rct_power has state class total_increasing, but its state is not strictly increasing. Triggered by state 164566.3 (164578.4) with last_updated set to 2024-05-22T11:14:35.393267+00:00. Please create a bug report at https://github.com/weltenwort/home-assistant-rct-power-integration/issues
Entity sensor.rct_power_storage_consumer_energy_consumption_year from integration rct_power has state class total_increasing, but its state is not strictly increasing. Triggered by state 1705967.6 (1705979.5) with last_updated set to 2024-05-22T11:14:35.393424+00:00. Please create a bug report at https://github.com/weltenwort/home-assistant-rct-power-integration/issues
Entity sensor.rct_power_storage_consumer_energy_consumption_total from integration rct_power has state class total_increasing, but its state is not strictly increasing. Triggered by state 3214125.5 (3214137.5) with last_updated set to 2024-05-22T11:14:35.393567+00:00. Please create a bug report at https://github.com/weltenwort/home-assistant-rct-power-integration/issues

brlnr23, 2024-09-08:
i just experienced the same (lucky me - only for about 30m). A restart fixed it... but it's strange. Recently an electrician told me, that multiple devices connected to RCT might lead to unwanted behavior (he told my to close my app while he was setting up stuff with his app). That might be a reason for this? When the error occurred for me, I had the app open several times and used it for looking up things. Maybe that caused the connection error ...

weltenwort, 2025-02-16:
Yes, unfortunately the inverter's network stack doesn't handle multiple concurrent connections well. This would have to be fixed by RCT in the firmware. I was considering creating a multiplexing proxy that disguises multiple devices as one, but haven't investigated it more thoroughly.

weltenwort, 2025-02-16:
Please re-open when you find any clues that this network problem might have originated with the integration.

closed 2025-02-16 by weltenwort