weltenwort/home-assistant-rct-power-integration #397: Deprecation warning (2024.09.1): async_forward_entry_setup will stop working in 2025.6

brlnr23, 2024-09-08:
still some time left, but since it hasn't been reported yet, here you go :) 

Logger: homeassistant.helpers.frame
Quelle: helpers/frame.py:151
Erstmals aufgetreten: 12:39:40 (1 Vorkommnisse)
Zuletzt protokolliert: 12:39:40

Detected code that calls async_forward_entry_setup for integration, rct_power with title: RCT Power Inverter at  and entry_id:, which is deprecated and will stop working in Home Assistant 2025.6, await async_forward_entry_setups instead. Please report this issue.

f6z2vfupg3ig, 2024-12-29:
Equal to to https://github.com/weltenwort/home-assistant-rct-power-integration/issues/397#issue-2512317742 but at different line number:

Logger: homeassistant.helpers.frame
Source: helpers/frame.py:234
First occurred: 5:08:50 PM (1 occurrences)
Last logged: 5:08:50 PM

Detected code that calls async_forward_entry_setup for integration, rct_power with title: RCT Power Inverter at {removed} and entry_id: {removed}, which is deprecated, await async_forward_entry_setups instead. This will stop working in Home Assistant 2025.6, please report this issue

weltenwort, 2025-02-16:
✅ This should have been fixed in [v0.14.1](https://github.com/weltenwort/home-assistant-rct-power-integration/releases/tag/v0.14.1).

💟 Thanks a lot for reporting!

closed 2025-02-16 by weltenwort