weltenwort/home-assistant-rct-power-integration #402: Updates to latest revisions leads to non working PV-Erzeugung Dashboard

christianmhuber, 2024-12-31:
@weltenwort Thanks again for the great integration. And since time is right, a happy new year!

Regarding my used hardware please refer to:
https://github.com/weltenwort/home-assistant-rct-power-integration/issues/323
Basically two Power Storage DC, DC 10 as Master, DC6 as Slave. Battery connected to DC10.

Energy Dashboard is using for "PV-Erzeugung"
- DC 10 Generator A Energy Production Total
- DC 10 Generator B Energy Production Total
- DC 6 Generator A Energy Production Total
- DC 6 Generator B Energy Production Total

I was still running my Home Assistant up to now at 2023.8.4.
For integration of a new device I had to make now some updates.

- Home Assistant Core 2023.8.4 to 2024.12.5
- Integration v0.10.0 to v0.13.1.
- HACS 1.34.0 to 2.0.1.

The upgrades left me at a non working "PV-Erzeugung" dashboard.
E.g. yesterday 30.12.2024.
[image]

Interestingly I got once a reading (there was now HA reboot yesterday), though only for one Power Storage DC 10, DC 6 missing completely. Basically at a restart of HA the energy dashboard will be updated ones (with all missed energy), then never again.

Data from the Power Storage DC are up to date at Home Assistant.
[image]

Somehow it looks similar to #323.

Any thoughts how I could get the energy dashboard working again?

Best regards
Christian

Logger: homeassistant
Quelle: components/sensor/__init__.py:690
Erstmals aufgetreten: 11:33:55 (1846 Vorkommnisse)
Zuletzt protokolliert: 18:48:48

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 267, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 478, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 177, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/config/custom_components/rct_power/lib/multi_coordinator_entity.py", line 34, in _handle_coordinator_update
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 690, in state
    raise ValueError(
    ...<5 lines>...
    )
ValueError: Sensor sensor.rct_power_storage_dc_6_external_energy_production_year has device class 'energy', state class 'total_increasing' unit 'Wh' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-finite value: 'nan'

[image]

oldskool00815, 2025-01-08:
Hello all, i also updated to the lastest version and the HA is not able to load the RCT Integration. 
Anyone else with the same problem?

speedy555, 2025-01-08:
I installed 2025.1.1 today and the integration is running well. Others also commented in already closed issue #403 that it is working.

Web-Spyder, 2025-02-18:
> Hello all, i also updated to the lastest version and the HA is not able to load the RCT Integration. Anyone else with the same problem?

I have sometimes issues with the integration and have solved it for me, deactivating (not deleteing!) the integration and immediately activating the integration again. Don't know why after reboots sometimes the integration ist not loaded correctly.

christianmhuber, 2025-02-19:
Hi Web-Spyder,
try to keep all three of them to the latest version:

- Home Assistant Core
- The integration itself and
- HACS

At least once the integration could not be loaded in my case having not all three up to date.

Best regards

Web-Spyder, 2025-03-12:
> Hi Web-Spyder, try to keep all three of them to the latest version:
> 
> * Home Assistant Core
> * The integration itself and
> * HACS
> 
> At least once the integration could not be loaded in my case having not all three up to date.
> 
> Best regards

It is working again.