weltenwort/home-assistant-rct-power-integration #403: Error after installation 2025.1

ifeelhorst, 2025-01-06:
## Version of the custom_component
version 0.10.0

## Configuration

```yaml
Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 5. Januar 2025 um 12:42:05 (1 Vorkommnisse)
Zuletzt protokolliert: 5. Januar 2025 um 12:42:05

Setup failed for custom integration 'rct_power': Unable to import component: cannot import name 'ELECTRIC_CURRENT_AMPERE' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/rct_power/__init__.py", line 25, in <module>
    from .lib.entities import all_entity_descriptions
  File "/config/custom_components/rct_power/lib/entities.py", line 7, in <module>
    from .device_info_helpers import get_battery_device_info, get_inverter_device_info
  File "/config/custom_components/rct_power/lib/device_info_helpers.py", line 4, in <module>
    from .entity import RctPowerEntity
  File "/config/custom_components/rct_power/lib/entity.py", line 21, in <module>
    from .device_class_helpers import guess_device_class_from_unit
  File "/config/custom_components/rct_power/lib/device_class_helpers.py", line 1, in <module>
    from homeassistant.const import (
    ...<12 lines>...
    )
ImportError: cannot import name 'ELECTRIC_CURRENT_AMPERE' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/rct_power/__init__.py", line 25, in <module>
    from .lib.entities import all_entity_descriptions
  File "/config/custom_components/rct_power/lib/entities.py", line 7, in <module>
    from .device_info_helpers import get_battery_device_info, get_inverter_device_info
  File "/config/custom_components/rct_power/lib/device_info_helpers.py", line 4, in <module>
    from .entity import RctPowerEntity
  File "/config/custom_components/rct_power/lib/entity.py", line 21, in <module>
    from .device_class_helpers import guess_device_class_from_unit
  File "/config/custom_components/rct_power/lib/device_class_helpers.py", line 1, in <module>
    from homeassistant.const import (
    ...<12 lines>...
    )
ImportError: cannot import name 'ELECTRIC_CURRENT_AMPERE' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)
```

## Describe the bug

After updating my system to version 2025.1 the integration won't start anymore. Along the error above there is a deprecated warning (see below)

Thanks for your help.

## Debug log

```
Logger: homeassistant.core
Quelle: helpers/deprecation.py:222
Erstmals aufgetreten: 5. Januar 2025 um 12:42:05 (2 Vorkommnisse)
Zuletzt protokolliert: 5. Januar 2025 um 12:42:05

Config was used from rct_power, this is a deprecated alias which will be removed in HA Core 2025.11. Use homeassistant.core_config.Config instead, please report it to the author of the 'rct_power' custom integration
Config was used from rct_power, this is a deprecated alias which will be removed in HA Core 2025.11. Use homeassistant.core_config.Config instead, please create a bug report at https://github.com/weltenwort/home-assistant-rct-power-integration/issues

```

speedy555, 2025-01-07:
Hi,
has anyone else problems with version 2025.1?
I haven't updated yet because I am afaraid of having the same problem.

gt026328, 2025-01-07:
Running on 2025.1 without any problems.Core				2025.1.1Supervisor			2024.12.3Operating System		14.1Frontend				20250106.0

CampinoK, 2025-01-07:
Running on HA 2025.1.0 with rct 0.13.1 since 2 days.
No Problems so far.

ifeelhorst, 2025-01-07:
> @ifeelhorst Which version installed? Version 0.10.0? There is a newer version available.

sorry I have also 0.13.1 

But @CampinoK has no problems! So it must be my setting. I will close the issue.

ifeelhorst, 2025-01-07:
it seems that I'm the only one with that issue, so it should be an internal problem with my setting.

closed 2025-01-07 by ifeelhorst