weltenwort/home-assistant-rct-power-integration #407: 3 RCT Sensor Entities gone missing in Homeassistant Energy dashboard

rpdelaat, 2025-01-13:
```text

Add your logs here. 

first message related to RCT in log was 
025-01-08 17:51:27.827 WARNING (ImportExecutor_0) [homeassistant.core] Config was used from rct_power, this is a deprecated alias which will be removed in HA Core 2025.11. Use homeassistant.core_config.Config instead, please report it to the author of the 'rct_power' custom integration

- I have 253 log messages since this first one:
- a series of messages are about an automation where I use  states.sensor.rct_power_storage_consumer_power.state.

2025-01-10 19:41:05.449 WARNING (ImportExecutor_0) [homeassistant.core] Config was used from rct_power, this is a deprecated alias which will be removed in HA Core 2025.11. Use homeassistant.core_config.Config instead, please report it to the author of the 'rct_power' custom integration
2025-01-10 19:41:19.753 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'rct_power' calls `async_add_job`, which should be reviewed against https://developers.home-assistant.io/blog/2024/03/13/deprecate_add_run_job for replacement options at custom_components/rct_power/__init__.py, line 122: hass.async_add_job(  # type: ignore. This will stop working in Home Assistant 2025.4, please create a bug report at https://github.com/weltenwort/home-assistant-rct-power-integration/issues
2025-01-10 19:41:19.758 WARNING (MainThread) [homeassistant.helpers.frame] Detected code that calls async_forward_entry_setup for integration, rct_power with title: RCT at 192.168.1.251:8899 and entry_id: 01JBWRXC7WH3TQSP4AHH8XNTB9, which is deprecated, await async_forward_entry_setups instead. This will stop working in Home Assistant 2025.6, please report this issue

I hope this is enough - first time I am reporting something like this.  Please let me know. 
```

weltenwort, 2025-02-16:
The warnings in the log entries should have been fixed in [v0.14.1](https://github.com/weltenwort/home-assistant-rct-power-integration/releases/tag/v0.14.1).

I'm not sure how it could relate to three entities missing. Could you share more details about the these entities?

rpdelaat, 2025-02-19:
Thanks for getting back to me. Here is what I mean in a short video with explanation. 
It seems I have the version 0.14.1 as you will see - and still I cannot find the entities mentioned in the documentation to put into my energy dashboard. Please have a look. https://xiber-technologies-gmbh.dubb.com/v/LkLerj

weltenwort, 2025-02-21:
Good to hear you found the cause 🎉 Thanks for reporting back!

closed 2025-02-21 by weltenwort