weltenwort/home-assistant-rct-power-integration #421: New Error by 0.14.1

phinotv, 2025-02-16:
This is the error message after installing 0.14.1 and restarting HA.

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.luxtronik2
Quelle: custom_components/luxtronik2/config_flow.py:269
Integration: Luxtronik
Erstmals aufgetreten: 20:23:32 (1 Vorkommnisse)
Zuletzt protokolliert: 20:23:32

Could not handle config_flow.async_step_dhcp DhcpServiceInfo(ip='192.168.123.111', hostname='ld7.rilke.internal', macaddress='000982212447')
Traceback (most recent call last):
  File "/config/custom_components/luxtronik2/config_flow.py", line 269, in async_step_dhcp
    self._abort_if_unique_id_configured()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2662, in _abort_if_unique_id_configured
    raise data_entry_flow.AbortFlow(error)
homeassistant.data_entry_flow.AbortFlow: Flow aborted: already_configured

[home-assistant_2025-02-16T19-26-49.115Z.log](https://github.com/user-attachments/files/18816010/home-assistant_2025-02-16T19-26-49.115Z.log)

CampinoK, 2025-02-16:
Luxtronik2 != Rctpower

cdce8p, 2025-02-16:
Are you sure this is caused by the `rct-power-integration`? Looks like an issue with `luxtronik2` instead.

weltenwort, 2025-02-17:
Sorry to hear you're running into errors in your Home Assistant installation. As the other community members pointed out, this doesn't seem to be related to this integration, though.

There seems to be a matching issue over here, though: https://github.com/BenPru/luxtronik/issues/309

The author apparently fixed this in the release [2025.1.4-Beta](https://github.com/BenPru/luxtronik/releases/tag/2025.1.4-Beta).

closed 2025-02-17 by weltenwort