weltenwort/home-assistant-rct-power-integration #461: Error when launching dev container

RuedigerMueller, 2025-03-30:
## Version of the custom_component

0.14.1

## Describe the bug
I forked the repository and then do a "Clone Repository in Container Volume". I accept the defaults from VSCode and VSCode starts building the container. At the very end when launching the container I get the below error. Not ruling out that something is wrong in my local setup as this is the first time working with devcontainers. However, the devcontainer provided by Home Assistent Core is building and running without any issues.

2025-03-30T18:24:19.422Z] Stop (1435 ms): Run: docker buildx build --load --build-context dev_containers_feature_content_source=/tmp/devcontainercli-root/container-features/0.74.0-1743359055606 --build-arg _DEV_CONTAINERS_BASE_IMAGE=mcr.microsoft.com/devcontainers/python:3.13 --build-arg _DEV_CONTAINERS_IMAGE_USER=root --build-arg _DEV_CONTAINERS_FEATURE_CONTENT_SOURCE=dev_container_feature_content_temp --target dev_containers_target_stage -f /tmp/devcontainercli-root/container-features/0.74.0-1743359055606/Dockerfile.extended -t vsc-home-assistant-rct-power-integration-bae45dc369b81bcc80b4251c92fa3bbc50e4458c843644bed239177cc43b7bfc-features /tmp/devcontainercli-root/empty-folder
[2025-03-30T18:24:19.423Z] Start: Run: docker events --format {{json .}} --filter event=start
[2025-03-30T18:24:19.424Z] Start: Starting container
[2025-03-30T18:24:19.424Z] Start: Run: docker run --sig-proxy=false -a STDOUT -a STDERR -p 8123:8123 --mount type=volume,src=vsc-remote-containers,dst=/workspaces --mount type=volume,src=vscode,dst=/vscode -l vsch.local.repository=https://github.com/RuedigerMueller/home-assistant-rct-power-integration.git -l vsch.local.repository.volume=vsc-remote-containers -l vsch.local.repository.folder=home-assistant-rct-power-integration -l devcontainer.config_file=/workspaces/home-assistant-rct-power-integration/.devcontainer/devcontainer.json -e HOME=/home/vscode -u vscode --userns=keep-id --entrypoint /bin/sh vsc-home-assistant-rct-power-integration-bae45dc369b81bcc80b4251c92fa3bbc50e4458c843644bed239177cc43b7bfc-features -c echo Container started
[2025-03-30T18:24:19.430Z] docker: --userns: invalid USER mode.
See 'docker run --help'.
[2025-03-30T18:24:19.431Z] Stop (7 ms): Run: docker run --sig-proxy=false -a STDOUT -a STDERR -p 8123:8123 --mount type=volume,src=vsc-remote-containers,dst=/workspaces --mount type=volume,src=vscode,dst=/vscode -l vsch.local.repository=https://github.com/RuedigerMueller/home-assistant-rct-power-integration.git -l vsch.local.repository.volume=vsc-remote-containers -l vsch.local.repository.folder=home-assistant-rct-power-integration -l devcontainer.config_file=/workspaces/home-assistant-rct-power-integration/.devcontainer/devcontainer.json -e HOME=/home/vscode -u vscode --userns=keep-id --entrypoint /bin/sh vsc-home-assistant-rct-power-integration-bae45dc369b81bcc80b4251c92fa3bbc50e4458c843644bed239177cc43b7bfc-features -c echo Container started
[2025-03-30T18:24:19.431Z] Start: Run: docker ps -q -a --filter label=vsch.local.repository=https://github.com/RuedigerMueller/home-assistant-rct-power-integration.git --filter label=vsch.local.repository.volume=vsc-remote-containers --filter label=vsch.local.repository.folder=home-assistant-rct-power-integration --filter label=devcontainer.config_file=/workspaces/home-assistant-rct-power-integration/.devcontainer/devcontainer.json
[2025-03-30T18:24:19.433Z] Stop (10 ms): Run: docker events --format {{json .}} --filter event=start
[2025-03-30T18:24:19.440Z] Stop (9 ms): Run: docker ps -q -a --filter label=vsch.local.repository=https://github.com/RuedigerMueller/home-assistant-rct-power-integration.git --filter label=vsch.local.repository.volume=vsc-remote-containers --filter label=vsch.local.repository.folder=home-assistant-rct-power-integration --filter label=devcontainer.config_file=/workspaces/home-assistant-rct-power-integration/.devcontainer/devcontainer.json
**[2025-03-30T18:24:19.441Z] Error: Command failed: docker run --sig-proxy=false -a STDOUT -a STDERR -p 8123:8123 --mount type=volume,src=vsc-remote-containers,dst=/workspaces --mount type=volume,src=vscode,dst=/vscode -l vsch.local.repository=https://github.com/RuedigerMueller/home-assistant-rct-power-integration.git -l vsch.local.repository.volume=vsc-remote-containers -l vsch.local.repository.folder=home-assistant-rct-power-integration -l devcontainer.config_file=/workspaces/home-assistant-rct-power-integration/.devcontainer/devcontainer.json -e HOME=/home/vscode -u vscode --userns=keep-id --entrypoint /bin/sh vsc-home-assistant-rct-power-integration-bae45dc369b81bcc80b4251c92fa3bbc50e4458c843644bed239177cc43b7bfc-features -c echo Container started**
[2025-03-30T18:24:19.441Z] trap "exit 0" 15
[2025-03-30T18:24:19.441Z] exec "$@"
[2025-03-30T18:24:19.441Z] while sleep 1 & wait $!; do :; done -
[2025-03-30T18:24:19.441Z]     at ytA (/root/.vscode-remote-containers/dist/dev-containers-cli-0.401.0/dist/spec-node/devContainersSpecCLI.js:468:1260)
[2025-03-30T18:24:19.441Z]     at bH (/root/.vscode-remote-containers/dist/dev-containers-cli-0.401.0/dist/spec-node/devContainersSpecCLI.js:468:1002)
[2025-03-30T18:24:19.442Z]     at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
[2025-03-30T18:24:19.442Z]     at async TtA (/root/.vscode-remote-containers/dist/dev-containers-cli-0.401.0/dist/spec-node/devContainersSpecCLI.js:485:3848)
[2025-03-30T18:24:19.442Z]     at async iB (/root/.vscode-remote-containers/dist/dev-containers-cli-0.401.0/dist/spec-node/devContainersSpecCLI.js:485:4963)
[2025-03-30T18:24:19.442Z]     at async wrA (/root/.vscode-remote-containers/dist/dev-containers-cli-0.401.0/dist/spec-node/devContainersSpecCLI.js:666:203)
[2025-03-30T18:24:19.442Z]     at async DrA (/root/.vscode-remote-containers/dist/dev-containers-cli-0.401.0/dist/spec-node/devContainersSpecCLI.js:665:14830)
[2025-03-30T18:24:19.442Z]     at async /root/.vscode-remote-containers/dist/dev-containers-cli-0.401.0/dist/spec-node/devContainersSpecCLI.js:485:1190
[2025-03-30T18:24:19.447Z] Stop (4661 ms): Run in container: node /root/.vscode-remote-containers/dist/dev-containers-cli-0.401.0/dist/spec-node/devContainersSpecCLI.js up --container-session-data-folder /tmp/devcontainers-5b348730-e39c-458f-a99e-a57905b198f71743359052001 --workspace-folder /workspaces/home-assistant-rct-power-integration --workspace-mount-consistency cached --gpu-availability detect --id-label vsch.local.repository=https://github.com/RuedigerMueller/home-assistant-rct-power-integration.git --id-label vsch.local.repository.volume=vsc-remote-containers --id-label vsch.local.repository.folder=home-assistant-rct-power-integration --id-label devcontainer.config_file=/workspaces/home-assistant-rct-power-integration/.devcontainer/devcontainer.json --log-level debug --log-format json --config /workspaces/home-assistant-rct-power-integration/.devcontainer/devcontainer.json --override-config /tmp/devcontainer-1acac65b-d6fd-4f0b-9487-e9b5a16cc9a0.json --default-user-env-probe loginInteractiveShell --mount type=volume,source=vsc-remote-containers,target=/workspaces,external=true --mount type=volume,source=vscode,target=/vscode,external=true --skip-post-create --update-remote-user-uid-default off --mount-workspace-git-root --include-configuration --include-merged-configuration
[2025-03-30T18:24:19.448Z] Exit code 1
[2025-03-30T18:24:19.448Z] Start: Run: docker rm -f 16d2251c54fb47e12387ce2c5411f0a919a89b97b4a26839f3996f667bc4c217
[2025-03-30T18:24:19.454Z] Command failed: node /root/.vscode-remote-containers/dist/dev-containers-cli-0.401.0/dist/spec-node/devContainersSpecCLI.js up --container-session-data-folder /tmp/devcontainers-5b348730-e39c-458f-a99e-a57905b198f71743359052001 --workspace-folder /workspaces/home-assistant-rct-power-integration --workspace-mount-consistency cached --gpu-availability detect --id-label vsch.local.repository=https://github.com/RuedigerMueller/home-assistant-rct-power-integration.git --id-label vsch.local.repository.volume=vsc-remote-containers --id-label vsch.local.repository.folder=home-assistant-rct-power-integration --id-label devcontainer.config_file=/workspaces/home-assistant-rct-power-integration/.devcontainer/devcontainer.json --log-level debug --log-format json --config /workspaces/home-assistant-rct-power-integration/.devcontainer/devcontainer.json --override-config /tmp/devcontainer-1acac65b-d6fd-4f0b-9487-e9b5a16cc9a0.json --default-user-env-probe loginInteractiveShell --mount type=volume,sourc [... cut ...] cc810d6b9066471b0b6fa75f557a15a1cbf31bb"}},{"Name":"runc","Version":"1.2.4","Details":{"GitCommit":"v1.2.4-0-g6c52b3f"}},{"Name":"docker-init","Version":"0.19.0","Details":{"GitCommit":"de40ad0"}}],"Version":"28.0.1","ApiVersion":"1.48","MinAPIVersion":"1.24","GitCommit":"bbd0a17","GoVersion":"go1.23.6","Os":"linux","Arch":"arm64","KernelVersion":"6.10.14-linuxkit","BuildTime":"2025-02-26T10:40:57.000000000+00:00"}}
[2025-03-30T18:25:25.727Z] Stop (20 ms): Run: docker version --format {{json .}}

weltenwort, 2025-04-05:
Hm, I suspect this might be connected to the addition of some extra dev container configuration I had to add to get this working with rootless podman, which I use instead of docker:

https://github.com/weltenwort/home-assistant-rct-power-integration/blob/4554ce75a772dfdba773b2a0dd6f243b0e61161d/.devcontainer/devcontainer.json#L33-L35

Does it work if you remove them?

RuedigerMueller, 2025-04-07:
Issue can be closed

closed 2025-04-07 by RuedigerMueller